rerun-io/rerun #4367: Visual glitch in a `ScrollArea` when two components are selected

abey79, 2023-11-28:
**Describe the bug**

When selecting two component and make the selection panel small, a visual glitch sometimes happen with the second component.

https://github.com/rerun-io/rerun/assets/49431240/07c1d243-35a9-4d8b-81eb-8d75dd245111

**To Reproduce**

Steps to reproduce the behavior:
1. detect_and_track example
2. select two components in the streams tree
3. reduce height of selection panel
4. scroll in selection panel

I've been able to reproduce in 0.8 and main as of today.

emilk, 2023-11-28:
Do we do any `set_clip_rect` shenanigans here?

Can we create a simple repro?

I wonder if this is related to https://github.com/emilk/egui/pull/2860 - does reverting it help?

abey79, 2023-11-28:
Playing with the glitch, I noticed that rows tend to appear _at once_ and not be clipped. This hints that the issue might be related with how the "rows to be displayed" is calculated, rather than a pixel clipping issue.

abey79, 2023-12-01:
Likely caused by https://github.com/emilk/egui/issues/3670

abey79, 2023-12-12:
Fixed by https://github.com/emilk/egui/pull/3690 and https://github.com/rerun-io/rerun/pull/4461

closed 2023-12-12 by abey79