rerun-io/rerun #4379: Add a way to link entity-paths across spaces and views such as for point-correspondences

jleibs, 2023-11-28:
Sometimes the same "thing" is logged at multiple entity-paths and multiple space views. In this case instance-ids should be considered 1:1 between the two paths.

We need a way to indicate to the UI that hovering or selecting one of these should also hover/select the corresponding entities.

This is very important to be able to show matched point-correspondences across 2D frames or between 2D and 3D spaces.

roym899, 2023-12-15:
Some thoughts on this after working on #4541 

This would also be nice to have for words / tokens in text (currently only text -> entity links exist), various slices in tensors (i.e., highlight one row for each point in a point set), etc.

It should also be possible to have multiple links for one entity. For example, one point could be linked to another point set, a row in a tensor, and a word in a text. All should be highlighted when hovering over any other. 

I think in some cases, it might be nice to allow setting the link direction.

demul, 2024-06-11:
any progress on this feature request?

Wumpf, 2024-06-11:
@demul Unfortunately not so far. There's various aspects to this and how ui interactions should play out is a bit hazy at this point. If you have any specific expectations on what this should look like please share, maybe there's some subset we could get to sooner