rerun-io/rerun #4415: Extremely slow data ingestion

emilk, 2023-12-01:
A user shared a puffin profile of _very_ slow data ingestion

[slow_ingestion.puffin.zip](https://github.com/rerun-io/rerun/files/13523843/slow_ingestion.puffin.zip)

[image]

[image]

What I see is this: the data ingestion (which is currently happening in the main rendering loop, see https://github.com/rerun-io/rerun/issues/4298) gets a message containing ~3800 rows of data. Processing that takes ~2 seconds, because each insertion takes a whopping ~9ms. The bulk of the insertion cost is re-sorting, which makes me suspect that the reason for this new slowdown is out-of-order insertions.

There are several layers of bad here:
* As mentioned, we shouldn't be doing insertions on the main render loop (https://github.com/rerun-io/rerun/issues/4298)
* If we do, we should make sure to not use more than a certain time budget
* Each insertion should not be this slow, even with out-of-order insertion

### Workarounds
* Run the logging with `RERUN_FLUSH_NUM_ROWS=1`: this will create smaller chunks of work for the viewer, which should get back the viewer frame-rate, but still with slow ingestion

* Stop logging data out-of-order. This could happen if you are using custom timelines that are not monotonically increasing. Rerun _should_ support out-of-order logging, but it seems like we're hitting a bad corner case here.

### Other notes
It would be great if we could tell from a puffin profile whether or not the code that was running was a debug or release build.
If this was from a debug build, then this whole issue is a mistake.

teh-cmc, 2023-12-01:
> It would be great if we could tell from a puffin profile whether or not the code that was running was a debug or release build.
If this was from a debug build, then this whole issue is a mistake.

+1 -- even the whole rerun versioning string ideally

emilk, 2023-12-04:
#4417 only pushed the problem to later:

[image]

The user was using `rec. set_time_sequence` temporarily for logging a few things, then never called `disable_timeline`. The user then went on to log 800k+ of other things, all ending up on the same timepoint in this "temporary" timeline.

emilk, 2024-01-05:
For future adventurers:

The problem with splitting the buckets further even when they share the same timepoint, is the definition of `IndexTable::buckets`:

```
pub buckets: BTreeMap<TimeInt, IndexedBucket>
```

Since `BTreeMap` doesn't allow duplicate keys, we would need to change the key there, or add "sub-buckets" inside `IndexedBucket` split on `RowId`. In either case, it is an annoying change.

PR #4417 (2023-12-01): Datastore: don't eagerly sort in bucket split rountine on ingestion path
This should fix #4415, although without knowing how exactly the user ended up in that situation it's hard to say for certain.

- #4415 

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Full build: [app.rerun.io](https://app.rerun.io/pr/4417/index.ht [... cut]
commit ec779889 (2023-12-02): Datastore: don't eagerly sort in bucket split rountine on ingestion path (#4417)

This should fix #4415, although without knowing how exactly the user
ended up in that situation it's hard to say for certain.
closed 2024-07-04 by teh-cmc