rerun-io/rerun #4426: Very large tables makes the UI sluggish

abey79, 2023-12-04:
Repro:

```python
import rerun as rr
from rerun.datatypes import ClassDescription

rr.init(args, "rerun_example_sluggish_ui", spawn=True)

annotation_context = rr.AnnotationContext(
    [
        ClassDescription(
            info=(0, "class_info", (255, 0, 0)),
            keypoint_annotations=[(i, f"keypoint {i}", (255, 255 - i, 0)) for i in range(65536)],
            keypoint_connections=[(i, 99 - i) for i in range(50)],
        ),
    ]
)

rr.log("annotation1", annotation_context)
```

https://github.com/rerun-io/rerun/assets/49431240/b882be39-4e22-45cc-bed5-b0f3d3c126b7

Using "smart" scroll area might not be the solution for when a large table is just one part of the UI, as is the case of AnnotationContext and the Selection Panel after https://github.com/rerun-io/rerun/pull/4416 is merged.

Maybe `egui_extras::Table` could have a fast short-circuit "is_visible" check in its drawing code?

emilk, 2023-12-04:
This sounds like a mis-use of `egui_extras::Table`. There are already fast ways to paint millions of rows: [rows](https://docs.rs/egui_extras/latest/egui_extras/struct.TableBody.html#method.rows)

abey79, 2023-12-04:
> This sounds like a mis-use of `egui_extras::Table`. There are already fast ways to paint millions of rows: [rows](https://docs.rs/egui_extras/latest/egui_extras/struct.TableBody.html#method.rows)

As implied in the description, I don't think this is a possible avenue if the table is "unrolled" in an outer ScrollArea, along with other UI elements.

abey79, 2023-12-04:
On the top of my mind, here is a list of steps that could possibly be taken to alleviate this issue. They are of course non-exclusive, and some have obvious added benefit. 
- don't unroll tables bigger than N
- show max N rows and a label "+ X rows..."
- show max N rows and "Show more..." button (some persistent state would keep the number of rows to be displayed)
- show max N rows and "doom-scroll"-style, auto-show-more (could be done by detecting when the rect for item N is getting close to the clip rect)
- add a search/filter box at the top of the table
- add pagination ("page 1 of X", next/prev buttons)
- open table in separate table view
- export table in CSV/parquet/xls/whatnot

emilk, 2023-12-05:
* Potential solution: https://github.com/emilk/egui/issues/3682

abey79, 2023-12-07:
This was actually fixed by #4416, which ended up migrating to `TableBody::rows()` everywhere. (This introduced significant display glitches that will shortly be resolved by the next `egui_extras` patch release that includes https://github.com/emilk/egui/pull/3690).

closed 2023-12-07 by abey79