rerun-io/rerun #4455: Crash when quickly resizing viewer with 3D view

roym899, 2023-12-07:
**Describe the bug**
When resizing the viewer for a little while it eventually crashes with the following error
```
2023-12-07T10:44:52Z ERROR wgpu::backend::direct] Handling wgpu errors as fatal by default

thread 'ThreadId(1)' panicked at 'wgpu error: Validation Error

Caused by:
    In a RenderPass
      note: encoder = `encoder`
    In a set_viewport command
    Viewport has invalid rect Rect { x: 224.0, y: 72.0, w: 932.0, h: 496.0 }; origin and/or size is less than or equal to 0, and/or is not contained in the render target Extent3d { width: 1055, height: 631, depth_or_array_layers: 1 }
```

https://github.com/rerun-io/rerun/assets/9785832/abb552f4-5d24-4d28-afc9-96fb777ef322

**To Reproduce**
Open the viewer and continuously resize for a few seconds. I think what matters is that it is fast (I guess too fast for my computer to handle something, wouldn't be surprised if this is harder to reproduce on better hardware or different OS). This is the code I use here, but I don't think it matters:
```
import numpy as np
import rerun as rr

rr.init("collapsing", spawn=True)
rr.log("parent/p1", rr.Points3D(np.random.rand(10,3)))
rr.log("parent/p2", rr.Points3D(np.random.rand(10,3)))
rr.log("parent/p3", rr.Points3D(np.random.rand(10,3)))
rr.log("parent/p4", rr.Points3D(np.random.rand(10,3)))
rr.log("parent/p5", rr.Points3D(np.random.rand(10,3)))
rr.log("parent/p6", rr.Points3D(np.random.rand(10,3)))

```

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Rerun version**
```
rerun_py 0.12.0-alpha.1+dev [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-unknown-linux-gnu
```

Wumpf, 2023-12-07:
We fixed the same error in egui prior to release of latest version. Unclear if it's still the same or a similar thing on our side (we do have a set_viewport call that could make the same mistake)

Wumpf, 2023-12-14:
confirmed that this no longer crashes, but only logs a (instant recoverable) error 🥳 
still should figure out though what keeps setting the wrong viewport on resize!

PR #5189 (2024-02-13): Fix (log only) error on quickly resizing the viewer
### What

* Fixes #4455

Got a good repro of this on my windows machine by using the screenshot action on a low-dpi screen (screenshot action will do a quick resize on native) while having the pose tracking example open.

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  *  [... cut]
commit 06c5cfbf (2024-02-13): Fix (log only) error on quickly resizing the viewer (#5189)

### What

* Fixes #4455

Got a good repro of this on my windows machine by using the screenshot
action on a low-dpi screen (screenshot action will do a quick resize on
native) while having the pose tracking example open.

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicab [... cut]
closed 2024-02-13 by Wumpf