rerun-io/rerun #4465: DisconnectSpace and TimeSeriesScalar

rangsjo, 2023-12-11:
**Describe the bug**

rerun_example_disconnect_space does not work if TimeSeriesScalar is logged in python.

**To Reproduce**
Steps to reproduce the behavior:
1. Run:

            # Slightly modified example code for disconnect space
            rr.init("rerun_example_disconnect_space", spawn=True)

            # These two points can be projected into the same space..
            rr.log("world/room1/point", rr.TimeSeriesScalar(0))
            rr.log("world/room2/point", rr.TimeSeriesScalar(1))

            # ..but this one lives in a completely separate space!
            rr.log("world/wormhole", rr.DisconnectedSpace())
            rr.log("world/wormhole/point", rr.TimeSeriesScalar(3))
3. Look in the viewer for disconnected spaces.

**Expected behavior**

Two different spaces should be shown, but as stated, all signals are shown in one single space.

**Screenshots**

[image]

**Desktop:**
 - OS: ubuntu 22.04 

**Rerun version**

rerun_py 0.10.1 [rustc 1.72.1 (d5c2e9c34 2023-09-13), LLVM 16.0.5] x86_64-unknown-linux-gnu release-0.10.1 e596e4e, built 2023-11-02T16:05:55Z

Wumpf, 2023-12-11:
Thanks for filing the issue! This is indeed a bug the way `DisconnectedSpace` is defined. That said, itself is a workaround for not being able to specify what goes into which space view from code which is luckily about to change! (we've been saying that for a while now, but we're finally almost there 😅 )
We're also meanwhile want to make the way space view heuristic spawning works a bit more scalable and predictable. As part of that we concluded that `DisconnectedSpace` will likely be renamed to `UndefinedTransform` and only be applicable to 2D & 3D space views. I.e. it is regarded as a special spatial transform.

.. but as things are this is a bug as described.

Wumpf, 2023-12-12:
Related to:

* #4388

emilk, 2024-01-03:
The plan is that setting up spaceviews should be done via the new blueprint API when this issue is done:
* https://github.com/rerun-io/rerun/issues/4167

PR #4935 (2024-01-30): `DisconnectedSpace` now only applies to spatial space views
### What

* Part of #4388
* Fixes #4465

This PR only addresses the documentation. `main` is actually in a slightly mixed state right now: Visualizability is already determined strictly with `DisconnectedSpace` only applying to spatial views, but we still have the legacy `SpaceInfo` used in some places that applies this concept to everything. This will be removed in a a follow-up.

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](htt [... cut]
commit 2b3e31bb (2024-01-30): `DisconnectedSpace` now only applies to spatial space views (#4935)

### What

* Part of #4388
* Fixes #4465

This PR only addresses the documentation. `main` is actually in a
slightly mixed state right now: Visualizability is already determined
strictly with `DisconnectedSpace` only applying to spatial views, but we
still have the legacy `SpaceInfo` used in some places that applies this
concept to everything. This will be removed in a a follow-up.

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Con [... cut]
closed 2024-01-30 by Wumpf