rerun-io/rerun #4466: Tracking issue: dataframe space view

abey79, 2023-12-11:
# Kinds

- [ ] Latest-at view (-> rows are entities/instances)
- [ ] Range query view (-> rows are event time)

# Features/toggle

- [ ] Expend/collapse entity instances
- [ ] Show/hide columns
- [ ] Show/hide indicator components
- [ ] Reorganise columns
- [ ] In "latest at" mode: "Last Updated" column

Wumpf, 2024-03-21:
It's not unlikely that we add some form of CSV support there as we go, but CSV is a quite opinionated and not well specified format (different locales may use . or , for decimal separators causing all sort of issues, different loaders allow all sort of column separation characters etc.), so this is probably best served with a custom data loader - already today you could easily write one that reads in a csv to a markdown and logs into Rerun or into a text log). Naturally, none of these representations is feature-wise all that great; for that I think we'd really need the feature described in this ticket.
To learn more about custom data loaders check the docs [here](https://www.rerun.io/docs/howto/open-any-file#adding-support-for-arbitrary-filetypes) and/or the respective examples, here's [the python one](https://github.com/rerun-io/rerun/tree/main/examples/python/external_data_loader).

gavrelina, 2024-08-05:
Updated some parts of the ticket based on design explorations documented here: https://github.com/rerun-io/rerun/issues/6957

nikolausWest, 2025-03-13:
This issue seems to not be kept up to date (some of the unchecked boxes I believe are actually implemented). @abey79 should this be closed in favor of a new tracking issue?

abey79, 2025-03-13:
I agree this is a bit all over the place and is not an actual representation of current priorities. Let's close this and reopen project-style issues as short term priorities are clarified.

closed 2025-03-13 by abey79