rerun-io/rerun #4486: Implement an entity path filter

emilk, 2023-12-12:
A set of include/exclude rules, with the most specific rule winning:

```
+ /world/**           # add everything…
- /world/roads/**     # …but remove all roads…
+ /world/roads/main   # …but show main road
```

`**` matches anything, including `/` _and parent_ (`/world/**` matches `/world`).

We will use this filter when describing what entities to include in a space view.

See https://www.notion.so/rerunio/Query-DSL-b9ebd276ff3f44bc91dc62f8b8857ea1?pvs=4  for more

abey79, 2023-12-12:
cc @martenbjork

closed 2023-12-21 by emilk

Guptajakala, 2025-04-07:
@emilk this is very useful, thank you!
Is there any plan to support wild cards at middle? E.g. `- world/roads/**/white`?