rerun-io/rerun #4491: Add breadcrumbs to the selection panel

abey79, 2023-12-12:
The enclosing context of the selected item should be displayed as breadcrumbs to facilitate the navigation. This issue is about the design and implementation of this feature.

#### Design issues

- [ ] address the difference/link between an entity from a space view query (aka `DataResult`) vs. entity in the streams
- [ ] address the duplication of the selected item:
      [image]
- [ ] have icon representing both type and sub-type for both containers and space view (and, ideally, query)

#### Implementation issues

- [ ] implement the capability to walk the hierarchy up
- [ ] implement the breadcrumb in selection panel

cc @martenbjork @jleibs

gavrelina, 2024-12-03:
Upd: we decided to use first entity letters instead of entity icons, as they give a better hint on what is the the actual name of the entity
and use a little store icon for the root in the streams

emilk, 2024-12-05:
* Most of it done in https://github.com/rerun-io/rerun/pull/8319

Still need to properly sync "projection" entities with the blueprint tree panel

jleibs, 2024-12-05:
> A) I don't know what the logic is for how they are displayed in the blueprint tree view

This happens here: https://github.com/rerun-io/rerun/blob/76d4663de1916bb6d1ee575b2224d848923a00f4/crates/viewer/re_blueprint_tree/src/blueprint_tree.rs#L390-L406

I'll reserve (B) for others to decide

>  (C) If it is a good model to follow, I don't know how to follow it

You can combine the snippet from above, with the logic to resolve a [`DataResult::Item`](https://github.com/rerun-io/rerun/blob/76d4663de1916bb6d1ee575b2224d848923a00f4/crates/viewer/re_viewer_context/src/item.rs#L32-L33)

Following the pattern here: https://github.com/rerun-io/rerun/blob/76d4663de1916bb6d1ee575b2224d848923a00f4/crates/viewer/re_selection_panel/src/selection_panel.rs#L261-L266

emilk, 2024-12-06:
Closed; next step in:
* https://github.com/rerun-io/rerun/issues/8342

closed 2024-12-06 by emilk