rerun-io/rerun #4497: Support loading `.ply` files

teh-cmc, 2023-12-12:
Support loading `.ply` files with the CLI and Viewer (not the SDKs!).

- Requires #4494
- Requires #4495 
- Requires #4496

Wumpf, 2023-12-12:
ply is generally a mesh/triangle format (it's literally the **Stanford Triangle Format**). However, it's worth noting that a lot of people also use it for point clouds! We need to re conciliate this somehow

teh-cmc, 2023-12-12:
I'm only focusing on point clouds for the builtin loader for, at least as a start (point cloud plys are a very common user request).

After which we can:
- expand on that builtin loader in the future to make it configurable etc, or
- just let users add their own ply loader now :)

teh-cmc, 2023-12-17:
Fixed at some point in the `DataLoader` PR series:
- #4516
- #4517 
- #4518 
- #4519 
- #4520 
- #4521 
- #4565
- #4566
- #4567

closed 2023-12-17 by teh-cmc