rerun-io/rerun #4500: Picking, point rendering and outlines spuriously don't work on some Chrome + OS version combinations

Wumpf, 2023-12-12:
Tracking for this issue
https://bugs.chromium.org/p/chromium/issues/detail?id=1510915

for all we know this affects any version where ANGLE directly translates to GL (usually against better judgement ;))

So far no success in isolating it to a simpler scene.

emilk, 2023-12-12:
According to @Wumpf wgpu is unaffected.

Sadly we cannot compile a single Wasm for both WebGL and WebGPU: https://github.com/gfx-rs/wgpu/issues/2804

Wumpf, 2023-12-12:
> According to @Wumpf wgpu is unaffected.

figuring this is a typo and supposed to mean `WebGPU`. Well, we don't have that good of a test coverage for WebGPU. For starters there's no WebGPU support on Chrome @ Linux to begin with. On Mac and Windows I have indeed not yet observed issues, but I didn't test with all the various Chromium derivatives that we're having trouble with here. Also, I recently only ran re_renderer demos and those worked well on WebGL as well, so not worth much

emilk, 2023-12-13:
I can confirm that WebGPU works in my MacOS Brave browser (running chromium 120.0.6099.71), but WebGL does not.

WebGL:
[image]

WebGPU:
[image]

Wumpf, 2024-01-09:
This got fixed in Chrome 120.0.6099.199
It affected all Chrome Angle renderers.

closed 2024-01-09 by Wumpf