rerun-io/rerun #4569: Move `re_ui` UI functions to a trait to be implemented by `egui::Ui`

abey79, 2023-12-18:
Most UI function in `re_ui` are `#[allow(clippy::unused_self)]`, but they take a `egui::Ui` argument (obviously). This could be more cleanly implemented as a trait that would be implemented for `egui::Ui`.

Something like:

```rust
pub trait ReUi {
  fn ui(&mut self) -> &mut egui::Ui;

  fn small_icon_button(&self, icon: &Icon) -> egui::Response {
        ui.add(
            egui::ImageButton::new(icon.as_image().fit_to_exact_size(Self::small_icon_size()))
                .tint(self.ui().visuals().widgets.inactive.fg_stroke.color),
        )
    }

    // etc.
}

impl ReUi for egui::Ui {
  fn ui(&mut self) -> &mut egui::Ui { self }
}
```

emilk, 2023-12-18:
be mindful that some of those functions are only NOT using self because they are using hard-coded values instead of the design tokens json

emilk, 2024-05-02:
I'm getting tired of passing around `ReUi` everywhere, so perhaps it's time to do this.

One thing though: `struct ReUi` currently store the `DesignTokens` it uses in some of its member functions.
We would need to store that somewhere else, e.g. in egui's data store: `ui.data(|data| data.get_temp::<Arc<DesignTokens>>())`

PR #6506 (2024-06-05): Remove `ReUi` and move its methods to `egui::{Ui|Context}` extension traits and to `DesignTokens`
### What

* Closes #4569 

This PR:
- Introduces `UiExt` and `ContextExt` extension traits for `egui::Ui`, resp. `egui::Context`.
- Refactor `DesignTokens` such that it's globally available via `re_ui::design_token()`.
- A new `re_ui::apply_style_and_install_loaders(&egui::Context)` function is introduced to initialise Rerun style in egui.
- `ReUi` is removed, and it's methods are dispatched to `DesignTokens`, `ContextExt`, and `UiExt`, depending on their nature.
- All `re_ui::ReUi` references are [... cut]
commit 5abbd297 (2024-06-05): Remove `ReUi` and move its methods to `egui::{Ui|Context}` extension traits and to `DesignTokens` (#6506)

### What

* Closes #4569 

This PR:
- Introduces `UiExt` and `ContextExt` extension traits for `egui::Ui`,
resp. `egui::Context`.
- Refactor `DesignTokens` such that it's globally available via
`re_ui::design_token()`.
- A new `re_ui::apply_style_and_install_loaders(&egui::Context)`
function is introduced to initialise Rerun style in egui.
- `ReUi` is removed, and it's methods are dispatched to `DesignTokens`,
`ContextExt`, and `UiExt`, depending on their nature.
- All `re_ui::ReUi` refer [... cut]
closed 2024-06-05 by abey79