rerun-io/rerun #4597: Add filtering to blueprint view

juliencbmeta, 2023-12-20:
**Is your feature request related to a problem? Please describe.**
When we have a long list in the blueprint view, it's harder to find the element you need. It would be nice to have some filtering. We can start by name. Ideally, we could also filter any other properties that would be attached each element in the list.

**Describe the solution you'd like**
Name filter in the blueprint view. Ability to filter by properties. But we would need a way to associate properties. In our case, I am thinking of a missing particular frame (for system that is expected to have 10 cameras, and 200 frames, maybe some of the cameras are missing a few frames, and we would like to be able to filter these in or out).

**Describe alternatives you've considered**
No alternative that I can think of

**Additional context**

abey79, 2023-12-20:
Thanks for the issue. Filtering by name is definitely something we plan to add in the blueprint tree (and elsewhere). Filtering by properties feels useful too.

The specific case of filtering by "missing data for the specific frame" is intriguing. It's not immediately obvious to me how that would happen exactly. I wonder if this would be better handled by range queries, either using the current "Visible Time Range" feature (or a minor extension of it), or the upcoming space view queries.

Pulling a bit on the "Visible Time Range" thread, would it be useful if that feature could be setup such that a camera space view would "blank" if the current frame is missing data?

juliencbmeta, 2023-12-20:
Would you mind pointing me to the "Visible Time Range" thread? Thanks!
In our use case, we will have more cameras to inspect than what can fit into the main viewer. For example, 100 cameras. We would need a quick way to find the cameras that have missing frames. Or to only look at cameras that have all their current frames available.

nikolausWest, 2023-12-20:
If what you are trying to do is find cameras with missing frames you can always try browsing through the timeline view in the bottom. You can see the individual events on the timeline so it should be easy to spot cameras that are missing frames there. I know that's not exactly the same as using that information to decide what camera streams to show in the main viewport

juliencbmeta, 2023-12-20:
Right, for that one, it sounds like it could work. But with over a 100 cameras, this visual check might take more time though.

abey79, 2023-12-20:
My thinking about "Visible Time Range" is that **IF** you could configure a frame interval such that data would be displayed only if it was actually logged in that interval, then you would observe blanking of the camera when on frame that has missing data.

This doesn't work now, because the Visible Time Range feature has the ["latest at" semantics](https://www.rerun.io/docs/getting-started/logging-rust#latest-at-semantics). In a nutshell, the "latest at" semantics means that data is considered "active" for as long as it's not overridden or explicitly cleared, so a narrow time range will return some data even if it was last logged _before_ the range interval. Namely, a camera space view configured with a single-frame visible interval would display the previous frame if the current one was missing.

Adding support for an alternative semantics that would display only data that has actually been _logged_ in the time range might be helpful in this case. We're tracking this discussion here: https://github.com/rerun-io/rerun/issues/4244