rerun-io/rerun #4614: Bounded message buffering for the web server

nikolausWest, 2023-12-22:
The web server currently buffers all incoming messages (to be able to serve web-viewers that connect late). This eventually leads to an out of memory crash.

We should make the size of the buffer configurable, and default to a reasonably small size such that the memory taken up by this buffer doesn't interfere with the memory used by the store for normal use on the same computer.

teh-cmc, 2023-12-22:
Re-sending state to recently connected clients is effectively the same as the `Save` feature in the viewer, isn't it? In which case we can just drop the buffer entirely and re-use that codepath.

That would be a good opportunity to finally address #1894 too.

emilk, 2024-01-04:
Closed by https://github.com/rerun-io/rerun/pull/4636

closed 2024-01-04 by emilk