rerun-io/rerun #4644: DepthImage backprojection doesn't set up correctly if the same entity is used for camera and image

kpreid, 2024-01-02:
Possibly the same sort of thing as #3296, but I can reproduce this on the latest code b4768f30b481a29b17dfbded685531bdd98f1286 .

**Describe the bug**
If a `DepthImage` and `Pinhole` are logged to the same entity, the backprojection will fail to render because the meter value is 0. Yet, it works fine if the value is manually changed.

**To Reproduce**
Steps to reproduce the behavior:
1. Modify `code-examples/depth_image_3d.rs` so that both `log()` operations use the same entity path.
2. `cargo run --bin rerun`
3. `cargo run --bin depth_image_3d`
4. Observe that no backprojected depth appears.
5. Select the 3D view of the entity and observe that "Backproject meter" is equal to 0.

**Expected behavior**
The logged meter value should be respected.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS Sonoma 14.2.1 (23C71)

**Rerun version**
rerun-cli 0.12.0-alpha.3 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-apple-darwin

Also reproduces with released rerun 0.11

kpreid, 2024-01-03:
Per discussion at https://github.com/rerun-io/rerun/pull/4627#issuecomment-1875694213 , maybe this sort of thing (multiple non-transform archetypes on the same path) isn't even supposed to be supported. But if so, there aren't really signposts against it, and #2568 specifically added support for a related case.

Wumpf, 2024-01-04:
> maybe this sort of thing (multiple non-transform archetypes on the same path) isn't even supposed to be supported

Generally, logging anything where components don't overwrite each other is fine. But we don't have a nice way of communicating that yet

Wumpf, 2024-01-04:
This is definitely not a case where components overlap, this should just work. But somehow the supplied meter is ignored. I'm investigating

PR #4679 (2024-01-05): Fix heuristic object properties being broken in some cases / fix DepthMeter being ignored sometimes
### What

* Fixes #4644 

The way we updated properties for auto-property values made it pretty much random (not literally, but feels as-if) when and when not we'd see updated `EditableAutoValue::Auto` values!

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have test [... cut]
commit 000a0ea5 (2024-01-05): Fix heuristic object properties being broken in some cases / fix DepthMeter being ignored sometimes (#4679)

### What

* Fixes #4644 

The way we updated properties for auto-property values made it pretty
much random (not literally, but feels as-if) when and when not we'd see
updated `EditableAutoValue::Auto` values!

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I  [... cut]
closed 2024-01-05 by Wumpf