rerun-io/rerun #4647: Don't use "Store ID" in the UI

nikolausWest, 2024-01-03:
When hovering a recording, in the recordings list, the on hover UI summary says "Store ID", which is an internal concept that should not be leaked to users. In this case it should instead be called "Recording ID", which matches the parameter `recording_id` in the SDKs.

[image]

RohanMishra315, 2024-01-05:
Hey @nikolausWest Can I take a look at it ? can you tell where Ui files are ?

emilk, 2024-01-08:
It's in `crates/re_viewer/src/ui/recordings_panel.rs`:

```
❯ rg "Store ID"
crates/re_viewer/src/ui/recordings_panel.rs
249:            re_ui.grid_left_hand_label(ui, "Store ID");
```

emilk, 2024-01-09:
I'm fixing this as part of https://github.com/rerun-io/rerun/issues/4649

closed 2024-01-10 by emilk