rerun-io/rerun #4655: Formalize specification/documentation of how transforms interact with heuristics and queries

jleibs, 2024-01-03:
There are several open questions at the moment regarding where transforms fit into the heuristic system.
 - Do they only impact "Spatial" views or can they be used to split content in different kinds of views.
 - Is there a mechanism to "query" entities up to the next transform?
 - Is there actually a difference between a "DisconnectedSpace" and a Null "SpatialTransform?"
 - Can a pinhole transform be converted to a rigid transform or does it create a permanent topological disconnect between 2d and 3d spaces.
 - Can 3D content be added to a 2D space without a transform?
 
This would improve clarity for users in issues like:
 - https://github.com/rerun-io/rerun/issues/4465

Wumpf, 2024-01-25:
We got answers for all of these by now and most of them even coded up:

* Do they only impact "Spatial" views or can they be used to split content in different kinds of views.
   * only impact spatial views. `SpaceInfo` should go away
* Is there a mechanism to "query" entities up to the next transform?
   * transforms propagation is part of general property propagation
* Is there actually a difference between a "DisconnectedSpace" and a Null "SpatialTransform?"
   * Yes. A disconnected space changes the spacial topology. A null spatial transform is handled as identity
* Can a pinhole transform be converted to a rigid transform or does it create a permanent topological disconnect between 2d and 3d spaces.
   * no. It's a permanent topological disconnect 
* Can 3D content be added to a 2D space without a transform?
   * no 

@jleibs plz check over these and close if we're on the same page. If not then this ticket is clearly not done 😄

jleibs, 2024-01-25:
Agree we're on the same page. Would be great to capture this somewhere durable. Almost like we need a spatial-transform architecture markdown doc.  (Which is where we would also add future notes on semantics for things like time-varying transforms https://github.com/rerun-io/rerun/issues/1522)

PR #4908 (2024-01-26): Notes on spatial transforms & spatial topology
### What

* Fixes #4655

As discussed on Slack, noted down what we decided on spatial transforms. Extended it a bit from there

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using newly built examples: [app.rerun.io](https://app.rerun.io/pr/{{pr.number}}/index.html)
  * Us [... cut]
commit 593fa495 (2024-01-26): Notes on spatial transforms & spatial topology (#4908)

### What

* Fixes #4655

As discussed on Slack, noted down what we decided on spatial transforms.
Extended it a bit from there

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[app.rerun.io](https://app.rerun.io/pr/{{pr.number}}/index.html) [... cut]
closed 2024-01-26 by Wumpf