rerun-io/rerun #4664: Scroll wheel zoom has extreme overshoot (on macOS)

kpreid, 2024-01-03:
**Describe the annoyance**
When zooming a 3D space view, it is easy to zoom in or out way too much — because the zoom continues for a very long time after I stop moving the scroll wheel. It *feels* like some layer of input processing is inappropriately applying touchpad-style flinging to my wheel input (which then works poorly because wheels don't report the _end_ of quick motion like a touchpad would).

Also, both wheel-based and touchpad-based zooming have a behavior where the rate of change of the displayed zoom seems to be held constant until it (apparently) catches up to an invisible target value. This is bad for precise control, because it means I can't predict when I've hit the desired target and stop my physical input motion there; I have to wait until it stops and then adjust. (Note that typical flinging implementations convey the distance that will be covered via a gradually decreasing velocity.)

It might be that the fling-ish behavior would be much more controllable if the constant-rate behavior was removed, but the combination is definitely An Annoyance and if it were up to me, I'd remove both.

**To Reproduce**
Steps to reproduce the behavior:
1. Get a macOS machine.
2. Attach a conventional USB wheel mouse.
3. Open a 3D space view.
4. Try to zoom to a desired scale precisely and quickly.

**Expected behavior**
Zoom tracks the motion of the wheel proportionally.

Actual behavior: After a movement of more than about 5 steps, zooming continues for about a second, often reducing the scene to a point if zooming out.

**Your goals**
Quickly scale data to a convenient viewing size (when auto-fitting doesn't work).

**Screenshots**
In this recording, I am making only brief motions of the scroll wheel but, if they are not very small, zooming continues for around a second afterward, with no hint as to when it is going to stop:

https://github.com/rerun-io/rerun/assets/779501/76ca03fc-05ba-48e9-81cf-6fc380f9aa09

**Desktop (please complete the following information):**
 - OS: macOS Sonoma 14.2.1 (23C71)
 - Affects rerun versions from 0.11.0 to b4768f30b481a29b17dfbded685531bdd98f1286

emilk, 2024-01-04:
We apply a smoothing to the scroll input that causes this latency. I'll take a quick stab at fixing it.

emilk, 2024-01-04:
Fix in https://github.com/rerun-io/rerun/pull/4668

kpreid, 2024-01-04:
I'd personally prefer even more direct control, but it's much better now. Thanks!