rerun-io/rerun #4669: Make space view `display_name` optional

abey79, 2024-01-04:
We currently have some kind of heuristic to provide an initial value for space view names, that roughly is:
1. `"$ENTITY_NAME"`, when created from heuristic with a given, non-root space origin
2. `"/ ($SPACE_VIEW_TYPE)"`, when the space origin is /
3. `"/ (empty $SPACE_VIEW_TYPE)"`, when creating an empty space view

This is inconsistant, and bit-rots extremely quickly when the use starts editing the space view's content. Plus, (1) contributes to the beginners confusion that "space view ~= entity" in the blueprint tree view.

I propose to make that field `Option`al and update the UI so that some subdued text is displayed until the user actually sets a custom name.

The actual text could further be `$SPACE_VIEW_TYPE view of $ENTITY_LAST_SEGMENT`, where entity could be either the space origin, or something derived from the query (if it is simple) e.g.:
- `2D view of /`
- `Text Log view of debug`
- `3D view of points`

If a suitable entity cannot be devised, the fallback would be `$SPACE_VIEW_TYPE view`, e.g. `Dataframe view`.

I further propose to forgo trying to stick a numbering scheme like we do currently (e.g. "/ (2D) (2)"). That numbering would have to be dynamic, which could cause confusion. Also, the synchronised hover/selection highlight should make it clear what corresponds to what. And if there really is confusion, then that's an encouragement for the user to name his views :)

nikolausWest, 2024-01-04:
Love it! Huge improvement

jleibs, 2024-01-04:
> if name clashes is a problem, then go ahead and name your space view

I don't have a strong enough opinion to fight for this either way, but I'll note that this runs contrary to most UX designs people are familiar with. Allowing things to have the same name is a recipe for confusion and mistakes.

The numbering doesn't have to be dynamic -- it could be based on something else like the last few digits of the space-view UUID.

I think the symmetric argument of: "If the ugly (2) bothers, you, then go ahead and name your space view" holds just as well.

closed 2024-01-17 by abey79