rerun-io/rerun #4689: Bad heuristics for objectron

jleibs, 2024-01-05:
**Describe the bug**
The heuristics are now creating a problematic series of space-views in the objectron example.

First there is a 2d "world" view.
Then there is a nested grid container with the expected 3D world view and 2D camera view.

**To Reproduce**
Go to: https://app.rerun.io/version/0.12.0-alpha.4/index.html

Load the objectron example.

**Screenshots**
[image]

emilk, 2024-01-05:
Hitting the "Re-populate" button on the blueprint fixes it, so it is something wrong with the incremental heuristics.

Running it locally also works correctly, likely because of timing issues (again, incremental heuristics might be a fault).

Wumpf, 2024-01-05:
can't repro this from the python script sadly. but that one has another small issue: camera frustum got really small

Wumpf, 2024-01-05:
We decided to tackle this likely pre-existing issue as part of 
* #4388

PR #4874 (2024-01-28): Improved automatic view creation heuristic, major speedup for scenes with many entities
### What

What space views get spawned is no longer determined by a centralized heuristic, but instead by one distributed to each space view class. Each of those heuristics can make much better use of existing information provided by store subscribers.

* Fixes most of #4388
   *  a few things on the tracking issue still left to investigate
* Fixes #3342 
   * more "obsoleted" arguably: We originally had a hack that 1d tensor logging would display a bar chart. That's not actually desirable. There is still t [... cut]
commit 15cd60d8 (2024-01-28): Improved automatic view creation heuristic, major speedup for scenes with many entities (#4874)

### What

What space views get spawned is no longer determined by a centralized
heuristic, but instead by one distributed to each space view class. Each
of those heuristics can make much better use of existing information
provided by store subscribers.

* Fixes most of #4388
   *  a few things on the tracking issue still left to investigate
* Fixes #3342 
* more "obsoleted" arguably: We originally had a hack that 1d tensor
logging would display a bar chart. That's not actually desirable. There
is s [... cut]
closed 2024-01-28 by Wumpf