rerun-io/rerun #4694: Viewer screenshots include the "image copied to clipboard" toast message

nikolausWest, 2024-01-05:
On `0.12.0a4` I can no longer get a clean screenshot of the viewer because the screenshot ends up including the message that a screenshot was taken.

[image]

It might be related to that a single screenshot produces three identical toast messages and three identical logs.

[image]

[image]

emilk, 2024-01-05:
How are you taking the screenshot? Seems like the command is triplicated

Wumpf, 2024-01-05:
it is. have a fix!

PR #4699 (2024-01-05): Fix screenshot command taking several screenshots at once
### What

* Fixes #4694

Regular screenshot!
[image]

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using newly built examples: [app.rerun.io](https://app.rerun.io/pr/4699/index.html)
  * Using examples from latest `main` build: [app.rerun.io](https://app.rerun. [... cut]
commit ce8675cf (2024-01-05): Fix screenshot command taking several screenshots at once (#4699)

### What

* Fixes #4694

Regular screenshot!

[image]

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[app.rerun.io](https://app.rerun.io/pr/4699/index.html)
* Using examples from latest `main` build:
[app.rerun.io](https://app. [... cut]
closed 2024-01-05 by Wumpf