rerun-io/rerun #4710: Prevent `scripts/run_all.py` from bit-rotting

abey79, 2024-01-05:
In #4146, I had to make a bunch of fixes to `run_all.py` to have a successful run. We should somehow try to avoid this script from bit-rotting.

abey79, 2024-01-05:
cc @jprochazk

jprochazk, 2024-01-08:
> We should somehow try to avoid this script from bit-rotting.

We'd have to run it on CI for maximum confidence, but we don't want to and/or can't right now. At least not with all the different possible combinations. Are there any other things we could do?

abey79, 2024-01-08:
> Are there any other things we could do?

Minimalistically, if we had some checklist of manual tests to run before a release that would include running the various `nox` session, it would at least catch any issue at that point. And making sure that our stuff runs on all supported Python version seems like a sane thing to be doing at least each release anyways.

emilk, 2024-04-10:
This script has significant overlap with `re_build_examples`.

I suggest we remove `run_all.py` and ports its functionality to `re_build_examples`.

abey79, 2024-04-10:
I took a quick look at this, and I'm no longer entirely sure it's worth re-implementing in rust. However, there are other ways to avoid bitrot, including running on the CI (at least night or maybe on push to main). That can be made easier with:
- #5898