rerun-io/rerun #4738: Unexpected stack trace when loading a file with no data-loader support.

jleibs, 2024-01-08:
Affects the cargo-installed binary, but not the pip installed binary

Installed from cargo:
```
$ rerun map.yaml
[2024-01-08T16:23:36Z INFO  re_data_source::load_file] Loading "map.yaml"…
Error: "map.yaml"

Caused by:
    No data-loader support for "map.yaml"

Stack backtrace:
   0: anyhow::context::<impl anyhow::Context<T,E> for core::result::Result<T,E>>::with_context
   1: re_data_source::data_source::DataSource::stream
   2: <core::iter::adapters::map::Map<I,F> as core::iter::traits::iterator::Iterator>::try_fold
   3: alloc::vec::in_place_collect::<impl alloc::vec::spec_from_iter::SpecFromIter<T,I> for alloc::vec::Vec<T>>::from_iter
   4: core::iter::adapters::try_process
   5: rerun::run::run_impl::{{closure}}
   6: tokio::runtime::park::CachedParkThread::block_on
   7: tokio::runtime::context::runtime::enter_runtime
   8: tokio::runtime::runtime::Runtime::block_on
   9: rerun::main
  10: std::sys_common::backtrace::__rust_begin_short_backtrace
  11: std::rt::lang_start::{{closure}}
  12: core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/core/src/ops/function.rs:284:13
  13: std::panicking::try::do_call
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/panicking.rs:500:40
  14: std::panicking::try
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/panicking.rs:464:19
  15: std::panic::catch_unwind
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/panic.rs:142:14
  16: std::rt::lang_start_internal::{{closure}}
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/rt.rs:148:48
  17: std::panicking::try::do_call
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/panicking.rs:500:40
  18: std::panicking::try
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/panicking.rs:464:19
  19: std::panic::catch_unwind
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/panic.rs:142:14
  20: std::rt::lang_start_internal
             at /rustc/d5c2e9c342b358556da91d61ed4133f6f50fc0c3/library/std/src/rt.rs:148:20
  21: main
  22: __libc_start_call_main
             at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
  23: __libc_start_main_impl
             at ./csu/../csu/libc-start.c:381:3
  24: _start
  ```
  
  Installed from pip:
  ```
  $ rerun map.yaml
[2024-01-08T16:25:14Z INFO  re_data_source::load_file] Loading "map.yaml"…
Traceback (most recent call last):
  File "/home/jleibs/rerun/venv/bin/rerun", line 8, in <module>
    sys.exit(main())
RuntimeError: "map.yaml" -> No data-loader support for "map.yaml"
  ```

emilk, 2024-01-09:
Interestingly, I can't reproduce from within the repository:

```
> cargo run -p rerun-cli --release --quiet -- map.yaml
[2024-01-09T09:25:43Z INFO  re_data_source::load_file] Loading "map.yaml"…
Error: "map.yaml"

Caused by:
    No data-loader support for "map.yaml"

>
```

emilk, 2024-01-09:
Hum, maybe it is an updated dependency that has added backtraces 🤔

emilk, 2024-01-09:
Ok, I have found it.

Starting in [`anyhow 1.0.77`](https://github.com/dtolnay/anyhow/releases/tag/1.0.77), a backtrace will be printed on error if `RUST_BACKTRACE` is set, and we set `RUST_BACKTRACE=1` in:

https://github.com/rerun-io/rerun/blob/d204936e4a186b6f264425f7a3d3849f94b7b86f/crates/re_log/src/setup.rs#L32-L35

When building the pip wheels, we use the `Cargo.lock` in our repository, with `anyhow 1.0.71`, so this is not an issue there.

closed 2024-01-09 by emilk