rerun-io/rerun #4756: Entities with `timeless=False` are logged twice when using custom logger

roym899, 2024-01-09:
**Describe the bug**
It seems that all entities with time are somehow logged twice when using a custom logger.

**To Reproduce**
1. Modify the docx example by changing to timeless=False.
2. Log the docx via drag-and-drop or `rerun example.docx`
3. Switch to `log_time`.

The text document appears twice on the timeline.

**Expected behavior**
I'd expect the same timeline as when running
```
import rerun as rr
rr.init("test", spawn=True)
rr.log("test", rr.TextDocument("test"), timeless=False)
```

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Rerun version**
```rerun_py 0.12.0-rc.6 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-unknown-linux-gnu release-0.12.0 4fc0c00, built 2024-01-09T14:41:10Z```

nikolausWest, 2024-01-09:
I can reproduce this as well:
[image]

teh-cmc, 2024-01-09:
And of course I can't on the other hand ><

```sh
$ RUST_LOG=off rerun example.docx

# Command Palette > Print data store

DataTable(17A8B8E0FEA5CF7661A0F4FA05E97B08):
┌───────────────────────────────────────────┬───────────────┬──────────────┬─────────────┬─────────────────┬─────────────────────────────────────────────────────────────────────────┬───────────────────────┐
│ RowId                                     ┆ EntityPath    ┆ NumInstances ┆ InstanceKey ┆ MediaType       ┆ Text                                                                    ┆ TextDocumentIndicator │
│ ---                                       ┆ ---           ┆ ---          ┆ ---         ┆ ---             ┆ ---                                                                     ┆ ---                   │
│ extension<rerun.controls.TUID>[struct[2]] ┆ str           ┆ u32          ┆ list[u64]   ┆ list[str]       ┆ list[str]                                                               ┆ list[null]            │
╞═══════════════════════════════════════════╪═══════════════╪══════════════╪═════════════╪═════════════════╪═════════════════════════════════════════════════════════════════════════╪═══════════════════════╡
│ 17A8B8E06ADD2A5F622D10964642120E          ┆ /example.docx ┆ 1            ┆ [0]         ┆ [text/markdown] ┆ [## My very important .docx                                             ┆ [-]                   │
│                                           ┆               ┆              ┆             ┆                 ┆                                                                         ┆                       │
│                                           ┆               ┆              ┆             ┆                 ┆ Lorem ipsum dolor sit amet, consectetur adipiscing elit, sed do eiusmo… ┆                       │
└───────────────────────────────────────────┴───────────────┴──────────────┴─────────────┴─────────────────┴─────────────────────────────────────────────────────────────────────────┴───────────────────────┘
```

nikolausWest, 2024-01-09:
I found the reason for my reproduction:
```bash
RUST_LOG=debug rerun example.docx
...
[2024-01-09T16:07:04Z DEBUG re_data_source::data_loader::loader_external] Loading data from filesystem using external loader… filepath="/Users/niko/Desktop/example.docx" loader="/Users/niko/.local/bin/rerun-loader-docx"
[2024-01-09T16:07:04Z DEBUG re_data_source::data_loader::loader_external] Loading data from filesystem using external loader… filepath="/Users/niko/Desktop/example.docx" loader="/Users/niko/src/rerun/rerun/venv11/bin/rerun-loader-docx"
```
There were two plugins running, one with each version.

PR #4762 (2024-01-09): Warn on duplicated data-loader in PATH

[image]

- Fixes #4756

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using newly built examples: [app.rerun.io](https://app.rerun.io/pr/4762/index.html)
  * Using examples from latest `main` build: [app.rerun.io](https://app.rerun.io/pr/4762/index.html?manifest_url=https://app.r [... cut]
commit fe1d94d2 (2024-01-09): Warn on duplicated data-loader in PATH (#4762)

[image]

- Fixes #4756

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[app.rerun.io](https://app.rerun.io/pr/4762/index.html)
* Using examples from latest `main` build:
[app.rerun.io](https://app.rerun.io/pr/4762/index.html?manifest_url=https://a [... cut]
commit 79850abe (2024-01-09): Warn on duplicated data-loader in PATH (#4762)

[image]

- Fixes #4756

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and
the [Code of
Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
* Using newly built examples:
[app.rerun.io](https://app.rerun.io/pr/4762/index.html)
* Using examples from latest `main` build:
[app.rerun.io](https://app.rerun.io/pr/4762/index.html?manifest_url=https://a [... cut]
closed 2024-01-09 by jleibs