rerun-io/rerun #4783: Developer dashboard

emilk, 2024-01-11:
I want a (public!) developer dashboard web page, for easily finding links and an overview of the state of things.

It should include links (and perhaps a summary of):

* Our benchmark tracking
* Our wasm/rrd size tracking
* The latest web app build
* CI-state of our repos (rerun, landing, …)
* Instructions on how to install the latest alpha
* Our app analytics
* Out website analytics
* …

emilk, 2024-04-24:
I consider this high priority.

[This PR](https://github.com/rerun-io/rerun/pull/5330) added 70 dependencies to `re_sdk`, and nobody noticed for TWO MONTHS _even though we had a bot post about it on the PR_.

If we had a developer dashboard that we would look at during standup or retro, this would not have slipped by.

Wumpf, 2024-04-24:
There's a posthog github action that could be useful for this if we decide not to dogfood on this https://github.com/PostHog/posthog-github-action

emilk, 2026-10-09:
We now have an internal tool for this (Rerun Pulse) that will soon cover most things

closed 2026-10-09 by emilk