rerun-io/rerun #4789: Fork `arrow2` and get rid of `polars`

emilk, 2024-01-11:
While we want to migrate from the `arrow2` crate to `arrow` (https://github.com/rerun-io/rerun/issues/3741), it is a big task that we would rather punt on right now. It is technical debt, but the debt is not going to grow signifnicantly.

One of the major reasons to migrate away from `arrow2` is because `DataType` has a huge overhead, especially when cloned.

We have a PR to fix it (https://github.com/jorgecarleitao/arrow2/pull/1469) but it is unmerged, because `arrow2` in unmaintained.

So: we fork `arrow2` as `re_arrow2`, merge our PR, and solve our immediate memory issue.

Since `polars`  require `arrow2`, we need to stop using it. We only have it for a few tests.

emilk, 2024-01-12:
One question is: what do we do with the tests and examples we have that depend on polars?

kylebarron, 2024-01-12:
fwiw, Polars also forked arrow2 into polars-arrow, and I don't think it has a dependency on arrow2 anymore

emilk, 2024-01-12:
We want to get rid of `polars` anyhow because of its huge dependency tree

emilk, 2024-01-15:
Blocked on landing caching to avoid conflicts

emilk, 2024-01-15:
Forking of arrow2 happening in https://github.com/rerun-io/arrow2/pull/2

I suggest we use it as `arrow2 = { package = "re_arrow2", version = "0.18" }`. That way we can keep using `arrow2` in the code for a smaller diff.

PR #4883 (2024-01-24): Datatype deduplication 2: switch to `re_arrow2`
Grunt work to switch to `re_arrow2` and all the breaking changes that come with it.

- Fixes #4789

---

Part of the tiny datatype deduplication PR series:
- #4880
- #4883 

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using newly built examples: [app.rerun.io](https://a [... cut]
commit c4dc7adb (2024-01-24): Datatype deduplication 2: switch to `re_arrow2` (#4883)

Grunt work to switch to `re_arrow2` and all the breaking changes that
come with it.

- Fixes #4789

---

Part of the tiny datatype deduplication PR series:
- #4880
- #4883
closed 2024-01-24 by teh-cmc