rerun-io/rerun #4794: Camera spins unevenly at <= 60 FPS rendering speed

roym899, 2024-01-12:
Once the displayed CPU time reaches ~= 10ms, the camera spins at uneven speed for me.

https://github.com/rerun-io/rerun/assets/9785832/e74b2825-828c-4f2b-be40-bb753093de09

**Rerun Version**
> rerun_py 0.12.0 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-unknown-linux-gnu release-0.12.0 afa112e, built 2024-01-09T17:56:39Z

**Desktop (please complete the following information):**
- Ubuntu 22.04

roym899, 2024-01-12:
Maybe also adding here that the same scene didn't have this issue back in 0.8 and that the spinning speed has increased noticeably since then.

https://github.com/rerun-io/rerun/assets/9785832/8a343624-5783-4727-9e64-1ebf91984f8b

Wumpf, 2024-01-12:
@emilk not sure this is a "quick issue". The code hat does the spinning looks like this:
```
            orbit_eye.rotate(egui::vec2(
                -response.ctx.input(|i| i.stable_dt).at_most(0.1) * 150.0,
                0.0,
            ));
```

so if high framerates cause issues here it implies that `stable_dt` isn't working properly

roym899, 2024-01-12:
I tried out some more versions and it seems like this has changed from 0.11 -> 0.12. 

Each frame takes ~2x as long, the rotation speed is higher, and the speed seems to be inconsistent.

emilk, 2024-01-16:
The docs explains it pretty well: https://docs.rs/egui/latest/egui/struct.InputState.html#structfield.stable_dt

emilk, 2024-01-17:
Ooh, it seems `stable_dt` is broken in `egui`! Will fix :)

emilk, 2024-01-17:
* Fix coming in https://github.com/emilk/egui/pull/3832

closed 2024-01-23 by emilk