rerun-io/rerun #4805: Asset rerun_cpp_sdk-0.12.0-multiplatform.zip is missing libraries for macOS and windows

rgolovanov, 2024-01-15:
**Describe the bug**
In assests of 0.12.0 or latest build the archive which suppose to contain multiplatform libraries misses libs for macOS and windows
see https://github.com/rerun-io/rerun/releases/tag/0.12.0

**To Reproduce**
Steps to reproduce the behavior:
1. Check asset [rerun_cpp_sdk-0.12.0-multiplatform.zip](https://github.com/rerun-io/rerun/releases/download/0.12.0/rerun_cpp_sdk-0.12.0-multiplatform.zip)

**Expected behavior**
All libraries in place

**Screenshots**
[image]
but all is in place for earlier releases:
[image]

**Rerun version**
0.12.0+

Wumpf, 2024-01-15:
thanks for reporting this! when testing the RC builds prior to the release the artifact was still correct, something must have broken in our CI. Will have a look asap!

Meanwhile you can use https://build.rerun.io/commit/4fc0c00/rerun_cpp_sdk.zip (which is from 0.12.0-rc6 which is fully compatible with 0.12.0)

Wumpf, 2024-01-15:
@rgolovanov I uploaded the correct artifacts now to the release 

https://github.com/rerun-io/rerun/releases/download/0.12.0/rerun_cpp_sdk.zip

Keeping this ticket open until we figured out why this happened and how to fix our release scripts

PR #4806 (2024-01-15): Execute PR-triggered cpp-upload only if there's no release label
### What

* (most likely) fixes  #4805

It seems what happened on 0.12 is that we first executed the regular release build which uploaded the correct C++ artifact https://github.com/rerun-io/rerun/actions/runs/7464893722/job/20313070683 and THEN ran the regular PR ci which only uploads a linux artifact, overwriting the previous https://github.com/rerun-io/rerun/actions/runs/7464905304/job/20313103967

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and  [... cut]
commit a36243bf (2024-01-15): Execute PR-triggered cpp-upload only if there's no release label (#4806)

### What

* (most likely) fixes  #4805

It seems what happened on 0.12 is that we first executed the regular
release build which uploaded the correct C++ artifact
https://github.com/rerun-io/rerun/actions/runs/7464893722/job/20313070683
and THEN ran the regular PR ci which only uploads a linux artifact,
overwriting the previous
https://github.com/rerun-io/rerun/actions/runs/7464905304/job/20313103967

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING [... cut]
commit 98809ea3 (2024-01-18): Execute PR-triggered cpp-upload only if there's no release label (#4806)

### What

* (most likely) fixes  #4805

It seems what happened on 0.12 is that we first executed the regular
release build which uploaded the correct C++ artifact
https://github.com/rerun-io/rerun/actions/runs/7464893722/job/20313070683
and THEN ran the regular PR ci which only uploads a linux artifact,
overwriting the previous
https://github.com/rerun-io/rerun/actions/runs/7464905304/job/20313103967

### Checklist
* [x] I have read and agree to [Contributor
Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING [... cut]
closed 2024-01-15 by Wumpf