rerun-io/rerun #4818: TimeSeriesView Archetype

nikolausWest, 2024-01-15:
Add a (blueprint) archetype for the TimeSeriesView to hold the relevant options.

Example options that should be settable per view that aren't currently:
- axis / zoom / grid options
  - [ ] add axis label
  - [ ] hide axis
  - [ ] secondary Y axis
  - [ ] set Y axis bounds
  - [ ] axis tick options
  - [ ] hide grid
  - [ ] grid options (min / max size?)
  - [ ] allow zoom / drag for x / y
- legend 
  - [ ] set custom name (instead of entity name)
  - [ ] ability to hide the legend

jleibs, 2024-01-18:
> allow zoom / drag for x / y

@nikolausWest can you elaborate on what you mean by this? We can already zoom and drag to pan x / y so I assume this is something different?

nikolausWest, 2024-01-19:
> can you elaborate on what you mean by this? We can already zoom and drag to pan x / y so I assume this is something different
I meant making those things optional. In the egui.rs plots demo you can turn zooming and panning on and of per axis.