rerun-io/rerun #4826: Add/remove entities modal doesnt allow adding to a 2D space view

abey79, 2024-01-16:
The add/remove entities modal doesn't let adding anything to a 2D space view with he ARKit example. This is possibly a heuristics bug?

[image]

### Repro

- Open ARKit example
- Remove everything from the blueprint
- Add an empty 2D space view
- Use the add/remove entities modal

Expected: stuff can be added (at least `camera_lowres` etc.)

Actual: nothing is enabled

abey79, 2024-01-16:
The reason for this is that that 2D space view has a space origin that is not compatible with any 2D content. It's space origin should be set to the entity that has the pinhole transform (or below) to be able to accept content.

For 0.12, the workaround is to create a 2d space view with a suitable space origin from the `+` button in the blueprint panel.

In 0.13, it will be possible to change the space origin of a space view.

Wumpf, 2024-01-28:
The behavior hasn't changed but:
* it is now possible to change the origin
* internally the situation is much better defined: None of the entities are visualizable from the space since they'd require a 3D->2D projection starting with a 2D eye which is an invalid transformation

So what's to do here now is to allow adding entities anyways (blueprint could do this!) and give the right hints why something doesn't execute a given visualizer.

PR #4937 (2024-01-30): Remove legacy SpaceInfo
### What

* Fixes #4388
   * this is the last piece 🥳  
* Related to
   * #4935
   * #4826

`SpaceInfo` used to be used to determine space connectivity, this is by now done by `SpatialTopology`

Not a "pure" refactor: Slight changes in behavior
* Adding a space view via the 'plus' top left uses the default-created list, not the now removed "all" list (which was poorly defined)
* Similarly, suggestions for root now use the default-created list
* adding entities via the add-ui no longer uses any kind of transform analysis (it relied on SpaceInfo) 
   * Does not fix  #4826 [... cut]
commit fb8d5d97 (2024-01-30): Remove legacy SpaceInfo (#4937)

### What

* Fixes #4388
   * this is the last piece 🥳  
* Related to
   * #4935
   * #4826

`SpaceInfo` used to be used to determine space connectivity, this is by
now done by `SpatialTopology`

Not a "pure" refactor: Slight changes in behavior
* Adding a space view via the 'plus' top left uses the default-created
list, not the now removed "all" list (which was poorly defined)
* Similarly, suggestions for root now use the default-created list
* adding entities via the add-ui no longer uses any kind of transform
analysis (it relied on SpaceInfo)
   * Does not fi [... cut]
closed 2024-01-30 by Wumpf