rerun-io/rerun #4845: Configurable lighting and shadows for 3D spatial scenes

jleibs, 2024-01-17:
Lighting, shading, and shadows can all have a dramatic impact on the ability to correctly interpret a rendering of 3D data.

It should be possible to do something like configure a light source and lighting properties via blueprint.

Related:
 - https://github.com/rerun-io/rerun/issues/1848
 - https://github.com/rerun-io/rerun/issues/1800

jviereck, 2025-01-13:
@Wumpf , we talked about this on Discord. I am interested to see how hard it would be to add shadows and light to rerun. Do you have an idea how to get started? I have basic understanding of OpenGL and did some rendering with WebGL but haven't worked with shadows and lights before.

Wumpf, 2025-01-13:
This touches on many different aspects of the renderer which we haven't built out yet and there is tons of tradeoffs and decisions that go into setting up systems for light configuration (and _even_ more so) for shadow casting. As much as I'd like to dive into that we don't have the bandwidth on the team right now to look into that.

jviereck, 2025-01-13:
Okay, no worries. I can understand if your team is working on other things.

If I come up with an idea for lights and shadows, can I write it up here and ask you to comment on them?

Wumpf, 2025-01-13:
Sure always happy about input, but don't expect any deep dives or committing to anything any time soon. This is definitely something we have to model in a more holistic manner and it's just not all that high up on the priority list right now.
Wgpu has a very basic shadow map example and Bevy implements some more sophisticated techniques if you want to start learning.
As I mentioned on Discord it may be very much worth first looking into (screen space) ambient occlusion since this requires less configuration and works in a wider variety of scenes out of the box (whereas lighting & shadow setup needs lots of user input on what the scene should look like and the various ways of implementing it have many trade-offs).

jviereck, 2025-01-25:
Would you be open to integrate a already existing 3D rendering engine like three-d? Three-d can render shadows [1].

[1]: https://github.com/asny/three-d/blob/master/examples/lighting/src/main.rs#L123

Wumpf, 2025-01-25:
At this point not really, no, at least not as a replacement for our primary built-in renderer (individual views being powered by some third party plugin etc. is a bit of a different story; it's something that comes up for plots every now and then. But extensibility isn't there either yet). We actually used three-d in an earlier prototype but we went with a custom render for a variety of reasons few of which have changed much, only that now switching renderer has a higher cost that back when we did. One of the primary advantages is that we can bend the datastructures to our exact needs and have high performance support on the platforms we care about.
We have the know-how inhouse to make shadow & lighting work for what we need and all of the imho hard problems I hinted at earlier around api/datastructure modelling & versatility across visualizations are pretty much the same.

jviereck, 2025-01-25:
Hi @Wumpf , thank you for your input on this. That makes sense.

jviereck, 2025-01-26:
Added a dedicated issue about implementing SSAO here: https://github.com/rerun-io/rerun/issues/8811