rerun-io/rerun #4866: Unix epoch timestamp display option; copy-able field

genemerewether, 2024-01-18:
Similar to https://github.com/rerun-io/rerun/issues/1714 - would it be possible to have a "unix epoch" timestamp display option? Sometimes external tools or code UIs use unix epoch seconds.

Also, is it possible to make the text there copy-able? Stretch-goal - make it a paste target that jumps the time to the pasted value.

E.g. `11677118942.733373` here on the timestamp panel:

[image]

emilk, 2024-01-19:
Sounds good! The easiest way to implement this would be to add a `UnixEpoch` as another variant in `enum TimeZone`.

Selectable labels (all labels) will come in the next major release of Rerun. Making it editable is a nice idea though.

MacOS, 2024-03-04:
Is someone working on that?

Wumpf, 2024-03-04:
@MacOS, no not right now, if you want to take a stab at it just go ahead! :)

commit 8e24b568 (2024-03-14): Add unix epoch as timezone option (#5450)

* Closes #4866

### What
Added a unix epoch timestamp display option, as mentioned in issue
[4866](https://github.com/rerun-io/rerun/issues/4866)

<img width="1470" alt="Screenshot 2024-03-10 at 23 25 16-min"
src="https://github.com/rerun-io/rerun/assets/21971561/c18ae1c1-9eb5-494a-bf30-78d2e0f7f363">

Not quite sure how compact time display should be handled: 

```diff
    /// Useful when showing dates/times on a timeline
    /// and you want it compact.
    ///
    /// Shows dates when zoomed out, shows times when zoomed in,
    /// shows relative m [... cut]