rerun-io/rerun #4899: Introduce space-root-relative paths for data-query expressions and results

jleibs, 2024-01-24:
Now that the space-view-root is runtime-configurable, it would be helpful if the default queries were relative to the root.

Current proposal for this is to use `./**` as the default query.

My proposal for how to proceed with this is:
 - Introduce a required `root` component on the DataQueryBlueprint
 - Use root when parsing the `EntityPathFilter` to convert to fully-qualified paths

Additionally we should track the set root in the QueryResult object and use that root instead of the currently adhoc `first_interesting_root` logic.

abey79, 2024-02-26:
Related issue re: proper visibility handling in the hierarchy: https://github.com/rerun-io/rerun/issues/3194

As noted in the discussion today, this is temporarily going to be more complicated when Groups are removed, and call for either:
- a complex, tristate sort of visibility widget in the blueprint tree;
- or a basic one, and advanced visibility controls added to the context menu.

abey79, 2024-02-27:
We discussed the "visibility" issue with @nikolausWest this morning and the outcome was:
- ok to regress in the short-term
- fixing #3194 would go _a long way_ towards making that regression acceptable as is
- risk: a larger redesign may end up being necessary to address the larger problem of property inheritance and override...
- ...but we can't really act on it right now. Let's move on according to the present issue and get a feel for it before further design.