rerun-io/rerun #4909: Present the root container as the top-level viewport container and differentiate it from other container

abey79, 2024-01-25:
TODO

nikolausWest, 2024-01-26:
Let's go with this approach. The alternative, which would be more similar to html or SwiftUI is to have an implicit vertical container at the root. However, I still think it's nice to have an explicit root to put global overrides on for instance so lets go with this

abey79, 2024-01-26:
Yes. If you want something else than a vertical container as viewport root, then you can set it and dont have to "manually" create a "sub-root" to achieve that effect.

nikolausWest, 2024-01-26:
Hmm... we actually need to think through how this interacts with the SDK. 

- Do we have a special Viewport container that you can use?
- What happens if I set up a hierarchy without a Viewport, do we wrap it in a viewport or convert the outermost container to a viewport?
  - If we convert it, that means I can never put that container in another container after the fact. If I've set overrides and options on that container that's kind of annoying.
  - On the other hand if we just document that the root container will represent the viewport and can't later be moved that might just be fine

abey79, 2024-01-26:
> Do we have a special Viewport container that you can use?

I don't think so. I'm not clear on the API details, but I'm anticipating that wherever we setup the viewport content, we can specify either one container (which will become the root), or one space view (which would then be the only viewport content, and will be added to an implicitly create root container).

> that means I can never put that container in another container after the fact

By definition of a root container (whether or not we give it a special treatment), we cannot move it anywhere, because a container may not be moved inside of itself.

You may move item _around_ (before/after) the root container, but performing that operation means implicitly creating a _new_ root container and adding both the newly moved item and the former root container in it. That's what we want to avoid here. 

> On the other hand if we just document that the root container will represent the viewport and can't later be moved that might just be fine

That's exactly the spirit of this issue.

PR #4989 (2024-02-01): Improve how the root container is displayed and handled in the blueprint tree
### What

This PR changes how the root container is displayed and handled:
- It's labeled as "Viewport", since it defines the top-level organisation of the viewport.
- It's not collapsible, so it doesn't have a triangle. This reduces the right-ward drift.
- It (still) cannot be dragged (but now the difference with other containers is more obvious). It now behaves better when dragged over though.
- It cannot be removed.

Other than that, it still behaves mostly as other containers (in particular, it can be selected an [... cut]
commit a1a4df28 (2024-02-01): Improve how the root container is displayed and handled in the blueprint tree (#4989)

### What

This PR changes how the root container is displayed and handled:
- It's labeled as "Viewport", since it defines the top-level
organisation of the viewport.
- It's not collapsible, so it doesn't have a triangle. This reduces the
right-ward drift.
- It (still) cannot be dragged (but now the difference with other
containers is more obvious). It now behaves better when dragged over
though.
- It cannot be removed.

Other than that, it still behaves mostly as other containers (in
particular, it can be se [... cut]
closed 2024-02-01 by abey79