rerun-io/rerun #4915: app.rerun.io won't load in chrome on linux

jleibs, 2024-01-25:
The egui update from https://github.com/rerun-io/rerun/pull/4885 appears to have caused a regression.

On attempted load, all I get is:
[image]
```
An error occurred during loading:
Failed to create wgpu adapter, no suitable adapter found.
Make sure you use a modern browser with either WebGL or WebGPU support enabled.
```

The problem does not appear to impact firefox, so something about chrome, specifically is confused.

emilk, 2024-01-26:
It works on Brave on macOS, chromium 121.0.6167.85

Wumpf, 2024-01-26:
Related to
* #4898 

According to log we try to run with WebGPU, but that's behind a feature flag on Linux and afaik not _actually_ really supported yet
https://github.com/gpuweb/gpuweb/wiki/Implementation-Status#chromium-chrome-edge-etc
Question now is ofc if you have this feature flag enabled, if not we should automatically use webgl on Linux Chrome unless explicitely specified otherwise.
Firefox works just nicely for you because in firefox (javascript) `navigator.gpu` is `undefined` thus we always pick webgl

emilk, 2024-01-26:
[image]

How can we read the flag in the client? Can we?

Wumpf, 2024-01-26:
If we can I don't know yet. The recommended way to check for webgpu support is to check `navigator.gpu`. Just repro'ed the issue as-is in a relatively fresh ubuntu VM with a freshly installed chrome. `navigator.gpu` is defined in the console just as expected (and kinda wrong) without setting any particular flags on the browser
I'll prod a bit around if we can do better than user-agent

Wumpf, 2024-01-26:
Looks like we can do an automatic retry with WebGL :)
Might need changes to egui though

emilk, 2024-01-26:
Automatic retry sounds awesome!

closed 2024-01-29 by emilk