rerun-io/rerun #4928: Add support for previewing blueprint changes in the viewport

abey79, 2024-01-29:
Whenever possible, the user should be able to preview the effect of their ongoing interaction.

For example:
- hovering a SV or container in the "Add SV/Container" modal should display it
- dragging a SV or container in the blueprint tree should display the effect of the candidate drop
- other? cc @nikolausWest 

This means we must have the ability to create a tile tree with temporary modifications without creating spurious  tree-to-blueprint synchronisations. Also, maybe add a `Preview`/`Commit` enum to relevant `TreeActions`. 

cc @jleibs

nikolausWest, 2024-01-29:
Suggested addition:

- Dragging a tile in the viewport should preview the change in the blueprint tree

abey79, 2024-01-29:
@nikolausWest I'm somewhat worried of the visual" ramifications of this one. Might be worth running it by @martenbjork

nikolausWest, 2024-01-29:
>  I'm somewhat worried of the "visual" ramifications of this one. 

Could you expand on your worries a bit @abey79?

abey79, 2024-01-29:
Two concerns:
1) Visual overload: dragging space views in the viewport might trigger fast-paced, possibly-wide-ranging tree diffs that would be difficult to parse for the user.
2) Hard-to-spot, "what changed?" tree diffs: for example, when displaying the candidate new space view when hovering in the "Add SV/Container" modal.

For (1), I suggest actually dropping that preview path. The desired end-result in the viewport, no point previewing stuff elsewhere when it's directly being edited.

For (2), some kind of highlight of the changed item.

nikolausWest, 2024-01-29:
Regarding (1) I actually don't think what you see in the viewport is enough. It's often unclear what the actual structural result will be when you drag a view around nested containers. However, I understand your point so we should probably think through the design a bit more before starting on this then