rerun-io/rerun #4931: CLI argument for setting max number of viewer threads

nikolausWest, 2024-01-29:
Since we started parallelizing the viewer, it can end up hogging too much CPU such that it interferes with other running processes.

Add a CLI argument that sets the max number of threads used by the viewer.

Wumpf, 2024-01-29:
`RAYON_NUM_THREADS` seems to be picked up since we're using the default rayon configuration

nikolausWest, 2024-01-29:
I still think we should expose it as a CLI argument (and ideally an argument to `rr.spawn` as well) for the purpose of making it easy to find for our users. Users shouldn't need to know we use Rayon

emilk, 2024-02-01:
This might be a good time for a `re_parallel` crate that handles the rayon initialization, so it is only initialized once.

Right now I think we don't initialize the thread pool explicitly when called from Python (meaning rayon threads are unnamed)

closed 2024-02-05 by emilk