rerun-io/rerun #4936: Resetting view by double clicking not working with huge amount of TimeSeries logs

seongs1024, 2024-01-29:
**Describe the bug**

**The view is not reset by double clicking** when it reaches to about 40k log events
(and a little bit of hanging about 5 seconds happened when I tried to brute-force multi-clicking to reset view)

**Screenshots**

[image]

[image]

**Desktop (please complete the following information):**
 - OS: macOS Catalina version 10.15.7

**Rerun version**

re_viewer: 0.12.1(bef0d78)
wasm32-unknown-unknown
rustc: 1.74.0
LLVM: 17.0.4

**Additional context**

abey79, 2024-01-29:
Could that be to the UI become too sluggish for double-click to be properly detected?

(BTW, the next release will _considerably_ improve the performance of timeseries).

seongs1024, 2024-01-29:
By the way, the memory panel consumes resources fairly a lot

teh-cmc, 2024-01-29:
Any chance you could try using a nightly build of the viewer @seongs1024 (either by building from source or through our [dev builds](https://github.com/rerun-io/rerun/releases/tag/prerelease))?

We're only a few days away from releasing `0.13`, which should drastically improve the performance of timeseries.
This is a perfect time to try it out and see if things work out for your use case.

Wumpf, 2024-01-29:
ah apologies, that's an intermittant issue we fixed by now, but we have to update a dependency for it (egui).
Thanks for linking to your project, we'll have look!

seongs1024, 2024-01-29:
Tested on `prerelease` 896ffcc by changing browser `Chrome` to `Safari`!

The performance enhancement is tremendously great!

But double click resetting view is still not working...

seongs1024, 2024-01-29:
I guess clicking issues are related when the UI becomes laggy.

Cannot click top left rerun icon to reset viewer when time goes

Wumpf, 2024-02-01:
@seongs1024 I tried your project on the `timeseries` branch but wasn't able to get the browser version running - Rerun currently has trouble building the web viewer outside of its own repo and I'm now really curious how you managed to 😄 .

So I changed it to use the native viewer and there everything works (unsurprisingly) smoothly on my machine. I suppose the native viewer is not an option for you for some reason? (Its performance is generally better by a huge margin, making the application a lot more responsive)

seongs1024, 2024-02-01:
FYI, there are steps to build my own web_viewer.

1. `git clone git@github.com:seongs1024/labs.git -b timeseries`
2. `cargo run --release`
3. _No `/re_run/web_viewer/re_viwer.js` error occurs_
4. _Go to the path where the compiler let you know like `${CARGO_HOME}/git/checkout/rerun-{hash}`_
5. _Run the script, it is in `scripts/setup_web.sh`, to install `binaryen` following [the Rerun docs](https://docs.rs/rerun/latest/rerun/)_
6. _Build rerun by `cargo build --release` in `${CARGO_HOME}/git/checkout/rerun-{hash}`_
7. Back to the the project directory what you cloned
8. Finally, `cargo run --release` again then it will show up on your browser!

Thank you! @Wumpf

teh-cmc, 2024-10-20:
Fixed by chunk-based timeseries

closed 2024-10-20 by teh-cmc