rerun-io/rerun #4939: Loading data with more than 1 space view deadlocks if `RAYON_NUM_THREADS=2`

teh-cmc, 2024-01-29:
This is true for both our latest release (`0.12.1`) and our latest `main`: `RAYON_NUM_THREADS=2 rerun plots.rrd` successfully starts a viewer and loads the data, but then deadlocks.

I think this might possibly hide another deadlock in the query cache, but we need to fix this one first to know for sure.

Putting on high priority as this might hide something very very wrong.

Wumpf, 2024-01-29:
Tried with both caching enabled and disabled, both from startup or changed at runtime and it doesn't repro for me so far

teh-cmc, 2024-01-29:
This is caused by external loaders, removing them all from my PATH fixes the issue -- something is blocking and/or waiting for another task on the compute threadpool?

No idea why the number of space views seemed to affect my earlier repro? We'll see what happens when I fix this.

PR #4942 (2024-01-29): External loader: don't do process IO on compute threadpool
Not quite sure what I was thinking when i wrote this :unamused: 

- Fixes #4938 
- Fixes #4939 

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicable)
* [x] I have tested the web demo (if applicable):
  * Using newly built examples: [app.rerun.io](https://app.rerun.io/pr/4942/index.html)
  * Using examples from latest `ma [... cut]
commit 43717831 (2024-01-29): External loader: don't do process IO on compute threadpool (#4942)

Not quite sure what I was thinking when i wrote this :unamused: 

- Fixes #4938 
- Fixes #4939
closed 2024-01-29 by teh-cmc