rerun-io/rerun #4947: Deadlock with primary caching is on

abey79, 2024-01-29:
I can consistently reproduce a dead-lock on main (as of 4371783) that happens only with primary caching on.

Repro step:
- reset the blueprint:
[image]

- left space view over middle one:
[image]

Deadlock occurs then.

Alternative repro:
- start with primary caching off
- enable primary caching

=> deadlock immediately occurs

Stack trace:

[image]

Repro RRD (800+MB, on private slack):
https://rerunio.slack.com/archives/C041NHU952S/p1706543181xxxx99?thread_ts=1706204242.101919&cid=C041NHU952S

abey79, 2024-01-30:
Another suspicious thread:
[image]

abey79, 2024-01-30:
And that error seems to popup shortly after the deadlock: 
[image]

abey79, 2024-01-30:
With `RAYON_NUM_THREADS=1` or `RAYON_NUM_THREADS=2`, the deadlock doesn't happen. With `RAYON_NUM_THREADS=3` it happens.

teh-cmc, 2024-01-30:
Reproduced together by duplicating a view a bunch of times

closed 2024-01-31 by teh-cmc