rerun-io/rerun #4962: Add/remove entity and query editor improvement

abey79, 2024-01-30:
This issue is about the minimal set of UI tweaks to bring the UI around Space View content (Add/remove entity, query dialog box, etc.) to a releasable state.

Always:
- move the "Clone SV" button to a different line, after the query editor if visible
- 

When the query DSL feature flag is OFF:
- back to "Add/remove entities" button label + adjust hover text accordingly

When the query DSL feature flag is ON:
- move "?" hover help to to field title hover text?

abey79, 2024-01-30:
@nikolausWest here is a proposal with mock-ups

https://www.figma.com/file/F02bljLYwuUy1ymlIn3Du0/Selection-panel-%2F-query-editor-%2F-release-0.13-fixup?type=whiteboard&node-id=0%3A1&t=KAckCAniXIOBhpl7-1

nikolausWest, 2024-01-30:
I added some comments (mainly copy) in the mockup but looks good!

PR #5022 (2024-02-02): Improve the UI for the entity query
### What

This PR implements the minimal sets of UI touch up to make the new query editor releasable.
**Note**: both commits best reviewed separately (first one is pure refactor)

* Fixes #4962

#### Query DSL feature flag off

[image]

[image]

#### Query DSL feature flag on

[image]

[image]

### Checklist
* [x] I have read and agree to [Contributor Guide](https://github.com/rerun-io/rerun/blob/main/CONTRIBUTING.md) and the [Code of Conduct](https://github.com/rerun-io/rerun/blob/main/CODE_OF_CONDUCT.md)
* [x] I've included a screenshot or gif (if applicab [... cut]
commit 7981b5d5 (2024-02-02): Improve the UI for the entity query (#5022)

### What

This PR implements the minimal sets of UI touch up to make the new query
editor releasable.
**Note**: both commits best reviewed separately (first one is pure
refactor)

* Fixes #4962

#### Query DSL feature flag off

<img width="322" alt="image"
src="https://github.com/rerun-io/rerun/assets/49431240/be2b6747-5254-41c6-874f-d2cd57e8fc83">

<img width="335" alt="image"
src="https://github.com/rerun-io/rerun/assets/49431240/255d7585-186b-4f9f-8b2d-4f572ac8063c">

#### Query DSL feature flag on

<img width="324" alt="image"
src="https://githu [... cut]
closed 2024-02-02 by abey79