rerun-io/rerun #4969: Plot becomes jagged with min-max aggregation

emilk, 2024-01-30:
min-max aggregation causes a zig-zag line to be generated, causing jagged look of smooth lines:

[image]

the solution is to instead output a thicker line, but that needs support in egui_plot

emilk, 2024-04-23:
We've had customers complain about the jagged lines, so perhaps the `MinMax` isn't a great default after all. It also seems pretty common to show a "simplified" (=potentially wrong) view of the data when you're zoomed out. Not that I think that's good, but maybe it's better than ugly looking plots.

emilk, 2026-09-22:
Related:
* https://github.com/emilk/egui/pull/8476
* https://github.com/rerun-io/reality/pull/1992

Wumpf, 2026-09-25:
Fixed (or at least substantially changed) in https://github.com/rerun-io/rerun/commit/12aad1b58c216340912c590631b955a05221ec06

closed 2026-09-25 by Wumpf