rerun-io/rerun #4996: Drag-n-drop issues with space view

teh-cmc, 2024-02-01:
Some issues I've encoutered while drag-n-dropping stuff around.

Much easier so screencast than to describe with words: see below.

#### Issue 1

https://github.com/rerun-io/rerun/assets/2910679/26e0e13d-562d-4520-82a5-2b6f7459aa76

#### Issue 2

https://github.com/rerun-io/rerun/assets/2910679/58591738-90c1-4ab7-92b5-581365d4e91b

emilk, 2024-02-05:
This is the text that pops up:

> [2024-02-05T14:08:04Z ERROR re_viewport::viewport] Visualizers for space view Placeholder("reconstruction") haven't been executed prior to display. This should never happen, please report a bug.

Wumpf, 2024-02-06:
The issue with (1) is likely that egui_tiles doens't report a view that we then unexpectedly draw: we first ask egui_tiles what views are going to be there, execute all visualizers for all spaces views in parallel, and _then_ draw egui_tiles. While doing so we rely on the results from the previously execueted visualizers. If that didn't happen that error shows up

closed 2024-02-06 by jleibs