rerun-io/rerun #5005: Store non-deterministically returns ComponentNotFound for new MarkerShape

jleibs, 2024-02-01:
This specifically impacts https://github.com/rerun-io/rerun/pull/5004

Can repro using the code:
```
import math

import rerun as rr

rr.init("rerun_example_scalar", spawn=True)

for step in range(0, 64):
    rr.set_time_sequence("step", step)
    rr.log("scalar/cos", rr.archetypes.Scalar(math.cos(step / 10.0)))
    rr.log("scalar/sin", rr.archetypes.Scalar(math.sin(step / 10.0)))
```

Sometimes everything works as expected:
[image]

But sometimes we get an error:
```
[2024-02-01T21:28:25Z ERROR re_viewport::system_execution] Error executing visualizer "SeriesPoint": Could not find component: rerun.components.MarkerShape
```
and the visualizer refuses to show the PointSeries:
[image]

Appears unrelated to the python logging since you can save an RRD file and reload it. Sometimes it will load properly, sometimes MarkerShape won't work.

jleibs, 2024-02-01:
This appears this is caching related as disabling primary_caching appears to avoid the issue.

teh-cmc, 2024-02-02:
I wouldn't be surprised if that's another bug I introduced in the reentrancy patch... :unamused:  looking.

closed 2024-02-02 by teh-cmc