rerun-io/rerun #5010: Cannot create a 2D from an albedo texture

teh-cmc, 2024-02-02:
See https://github.com/rerun-io/rerun/pull/4894#issuecomment-1923374263:
> The fact that we cannot create a 2D view of the albedo texture when inspecting it because it is not in itself an entity is an interesting and extremely frustrating limitation of our current system.
> 
> I'll probably open an issue actually...
> 
> ![image](https://private-user-images.githubusercontent.com/2910679/301789551-3317e4de-2393-4574-9fc7-6c1450f63276.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY4Njc1ODIsIm5iZiI6MTcwNjg2NzI4MiwicGF0aCI6Ii8yOTEwNjc5LzMwMTc4OTU1MS0zMzE3ZTRkZS0yMzkzLTQ1NzQtOWZjNy02YzE0NTBmNjMyNzYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDJUMDk0ODAyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YzkyZmRmODE0MDJlMzY2OGZkM2MxMDJjNWMxMTQwNDVjZjU4NzM5Y2FiNWRhMGQ4OTNmNjY3YmFhN2YzMmExZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.EAuDjp9CGffTaCM5IPrDc8Pf-PsjTccsLB6t_ZqEZZg)

Wumpf, 2024-02-03:
I think the main problematic aspects on this are right now:
* the image visualizer doesn't want to run because it doesn't see the the required indicator
   * (check on this!) it should be in the "applicable" list for that visualizer and in the "visualizable" list for any 2d space view created at the mesh
* ➡️  it doesn't want to show the entity in the tree / doesn't offer it 
* ➡️ ➡️ you can't even set the new visualizer override because it doesn't show up in the tree (and we haven't shown those yet)

further work on visualizer overrides, queries and blueprint ui needs to take this problem into account!

Wumpf, 2024-07-09:
[image]
0.17 can do this!

closed 2024-07-09 by Wumpf