rerun-io/rerun #5018: Endless ui refresh

emilk, 2024-02-02:
Something is calling `request_repaint` way too often.

* Could be https://github.com/emilk/egui/pull/3930
* Could be solved by https://github.com/emilk/egui/issues/3931

abey79, 2024-02-05:
repro on main:
- `just py-build`
- `python examples/python/face_tracking/main.py`
- let the script run for a few seconds then kill it
- reset viewport

=> no endless UI refresh

mess just a little bit with the space view:
- move the 3D space view tile around (tab / un-tab it)
- zoom/scroll  the 3D space view a little bit

After a few of those ops, the infinite UI refresh triggers.

abey79, 2024-02-05:
Still using the face tracking example, when setting the blueprint as follows and restarting, I can trigger the infinite refresh by just rotating/zooming the 3d view—no tile dragging needed.

[image]

jleibs, 2024-02-06:
This issue is still affecting head of main.  It doesn't impact all scenes, but for example in `arkit_scnes` the scene constant repaints.

emilk, 2024-02-06:
@jleibs you just need to uncheck this checkbox:

[image]

It is off by default, but you ran an earlier version on `main` where it was on by default, and that is now saved. Or just reset your Rerun.

In other words: this won't be a problem for our users.

emilk, 2024-02-06:
nope, wrong again

closed 2024-02-06 by emilk