rerun-io/rerun #5055: Prerelease install command not working when same version is already installed

roym899, 2024-02-06:
**Describe the annoyance**
The command to install the prerelease from here https://github.com/rerun-io/rerun/releases will not work if the same version is already installed.

**Possible solution**
Add `--force-reinstall` to the command?

emilk, 2024-02-06:
To clarify: This should only affect pre-releases where the version number is the same, so this is low priority.

roym899, 2024-02-06:
Adding that the command also does not work when there is a newer version (like a release candidate) on pypi already. It will install the release candidate then.

roym899, 2024-02-07:
Maybe something like this would work more reliably:
```
pip install --no-cache-dir --force-reinstall -f https://build.rerun.io/commit/fccd095/wheels --upgrade rerun-sdk==0.13.0a3
```