rerun-io/rerun #5058: Can't visualize Image once Pinhole is logged

roym899, 2024-02-06:
**Describe the bug**
In the current prerelease I can't seem to visualize 2D images once a Pinhole is logged (even if the paths are completely separate).

**To Reproduce**
```python
import rerun as rr
import numpy as np

rr.init("test", spawn=True)
rr.log("img1", rr.Image((256 * np.random.rand(100, 100, 3)).astype(np.uint8)))
rr.log("pinhole", rr.Pinhole(focal_length=100, resolution=(100, 100)))
```
In this case, the add space view button is completely unresponsive.

```python
import rerun as rr
import numpy as np

rr.init("test", spawn=True)
rr.log("points", rr.Points3D(np.random.rand(500,3)))
rr.log("img1", rr.Image(np.random.rand(100, 100, 3)))
rr.log("pinhole", rr.Pinhole(focal_length=100, resolution=(100, 100)))
```
In this case, the button works; however, it's not possible to add the image to the 2D space view.

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

**Rerun version**
> rerun_py 0.13.0-alpha.4 [rustc 1.74.0 (79e9716c9 2023-11-13), LLVM 17.0.4] x86_64-unknown-linux-gnu release-0.13.0-alpha.4 e7325d4, built 2024-02-06T10:51:01Z

Wumpf, 2024-02-06:
not entirely sure how it is expected to work. If it did before I'd say that was a bug?
The only thing that should display the image is a space view *with it's origin set to `img1`*. But generally we now assume that everything that isn't under `pinhole` is a 3d space and thus we can't walk any entities in a 2d view there.

Seems we need to make this logic a lot more transparent to the user

Wumpf, 2024-02-06:
[image]

works as described above. It's a bit annoying though that once I set the origin for the new 2d view I still have to edit the query, that seems wrong-ish to me

roym899, 2024-02-06:
Hm, I see. 

I guess for the first case it is definitely buggy that the add view button is not working. 

Ignoring that, it seems somewhat inconvenient to always have to manually adjust the space origin to visualize an image. Maybe an option could be to have no space origin as the default option for the 2D view in which case multiple images could simply be added with similar behavior as before.

closed 2024-02-06 by jleibs

Wumpf, 2024-02-07:
@roym899 FYI that space view is now created automatically for you