rerun-io/rerun #5070: The mesh is no longer visible in arkit_scenes

jleibs, 2024-02-06:
**Describe the bug**

The mesh no longer shows up.

Go to: 

**To Reproduce**
Go to: https://app.rerun.io/commit/b9cd458/?url=https://app.rerun.io/version/nightly/examples/arkit_scenes.rrd

**Expected behavior**
Should see the mesh

**Screenshots**
[image]

**Backtrace**

**Desktop (please complete the following information):**
 - OS: Ubuntu.
 - Reproduces on both Native and Web

**Rerun version**

**Additional context**

emilk, 2024-02-06:
Actually, the mesh is there. You just have to zoom in (on the labels) a lot:

[image]

It is the depth point cloud that is extremely large

emilk, 2024-02-06:
The problem sits in `SpaceViewBlueprint::on_frame_start`. It fills in a `per_system_entities` which is _empty_.

Why? Because every single entity has zero visualizers according to the `DataQueryResult` from `lookup_query_result`.

I suspect this got broken in https://github.com/rerun-io/rerun/pull/5050

jleibs, 2024-02-06:
Confirmed, fixed by: https://github.com/rerun-io/rerun/pull/5068

closed 2024-02-06 by emilk