rerun-io/rerun #5073: Intermittent crash on chrome

jleibs, 2024-02-06:
```
Rerun has crashed.

panicked at /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/wgpu-hal-0.19.1/src/gles/device.rs:752:52:
called `Result::unwrap()` on an `Err` value: "Unable to create texture object"
```

[image]

OS Version: Ubuntu 22.10 (Sway / Wayland)
Browser: Google Chrome Version 120.0.6099.199 (Official Build) (64-bit)

Console log:
```
index.html:240 loading wasm…
index.html:323 Wasm loaded. Starting app…
re_viewer.js:2813 [eframe::web::web_painter_wgpu] eframe-0.26.0/src/web/web_painter_wgpu.rs:88: Creating wgpu painter
re_viewer.js:2813 [eframe::web::web_painter_wgpu] eframe-0.26.0/src/web/web_painter_wgpu.rs:109: Creating wgpu instance with backends Backends(GL | BROWSER_WEBGPU)
re_viewer.js:2813 [eframe::web::web_painter_wgpu] eframe-0.26.0/src/web/web_painter_wgpu.rs:134: Attempting to create WebGPU adapter to check for support.
re_viewer.js:2120 WebGPU is experimental on this platform. See https://github.com/gpuweb/gpuweb/wiki/Implementation-Status#implementation-status
re_viewer.js:2120 Failed to create WebGPU Context Provider
imports.wbg.__wbg_requestAdapter_3148ca06e5f49220 @ re_viewer.js:2120
re_viewer.js:2813 [eframe::web::web_painter_wgpu] eframe-0.26.0/src/web/web_painter_wgpu.rs:149: Failed to create WebGPU adapter.
re_viewer.js:2813 [eframe::web::web_painter_wgpu] eframe-0.26.0/src/web/web_painter_wgpu.rs:152: Recreating wgpu instance with WebGL backend only.
re_viewer.js:2813 [egui_wgpu] egui-wgpu-0.26.0/src/lib.rs:131: Picked wgpu adapter: backend: Gl, device_type: IntegratedGpu, name: "ANGLE (Intel, Mesa Intel(R) Graphics (ADL GT2), OpenGL ES 3.2)", vendor: 0x8086
re_viewer.js:2813 [eframe::web::web_painter_wgpu] eframe-0.26.0/src/web/web_painter_wgpu.rs:187: wgpu painter initialized.
re_viewer.js:2807 [eframe::web::storage] eframe-0.26.0/src/web/storage.rs:23: Failed to parse memory RON: 1:51024: Expected a nonzero u64 but found the unsigned integer `0` instead
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2813 [re_renderer::context] re_renderer/src/context.rs:541: wgpu adapter backend: Gl, device_type: IntegratedGpu, name: "ANGLE (Intel, Mesa Intel(R) Graphics (ADL GT2), OpenGL ES 3.2)"
re_viewer.js:2813 [re_viewer::web] re_viewer/src/web.rs:62: Web app started.
index.html:368 App started.
re_viewer.js:2813 [egui::context] egui-0.26.0/src/context.rs:520: Loading new font definitions
re_viewer.js:2813 [re_log_encoding::stream_rrd_from_http] re_log_encoding/src/stream_rrd_from_http.rs:62: Downloading .rrd file from "https://app.rerun.io/commit/29e81e9/examples/structure_from_motion.rrd"…
re_viewer.js:2813 [re_log_encoding::stream_rrd_from_http] re_log_encoding/src/stream_rrd_from_http.rs:76: Decoding .rrd file from "https://app.rerun.io/commit/29e81e9/examples/structure_from_motion.rrd"…
2re_viewer.js:2813 [re_viewer::app] re_viewer/src/app.rs:965: Opening a new recording: StoreInfo { application_id: ApplicationId("rerun_example_structure_from_motion"), store_id: StoreId { kind: Recording, id: "ab67ae65-fa9a-4404-aad3-b9740cdae6a9" }, is_official_example: true, started: "17:27:18.113598Z", store_source: PythonSdk(PythonVersion { major: 3, minor: 11, patch: 6, suffix: "" }), store_kind: Recording }
re_viewer.js:2813 [re_log_encoding::stream_rrd_from_http] re_log_encoding/src/stream_rrd_from_http.rs:87: Finished decoding .rrd file from "https://app.rerun.io/commit/29e81e9/examples/structure_from_motion.rrd"…
re_viewer.js:2813 [re_viewer::app] re_viewer/src/app.rs:932: Data source http://https://app.rerun.io/commit/29e81e9/examples/structure_from_motion.rrd has left
re_viewer.js:2807 ERROR: [wgpu_core::device::global] wgpu-core-0.19.0/src/device/global.rs:1602: Device::create_render_pipeline error: Internal error in ShaderStages(VERTEX | FRAGMENT) shader: 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu::backend::wgpu_core] wgpu-0.19.1/src/backend/wgpu_core.rs:1152: Shader translation error for stage ShaderStages(VERTEX | FRAGMENT): 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu::backend::wgpu_core] wgpu-0.19.1/src/backend/wgpu_core.rs:1153: Please report it to https://github.com/gfx-rs/wgpu
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu_hal::gles::device] wgpu-hal-0.19.1/src/gles/device.rs:200: 	Shader compilation failed: 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu_core::device::global] wgpu-core-0.19.0/src/device/global.rs:1602: Device::create_render_pipeline error: Internal error in ShaderStages(VERTEX) shader: 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu::backend::wgpu_core] wgpu-0.19.1/src/backend/wgpu_core.rs:1152: Shader translation error for stage ShaderStages(VERTEX): 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu::backend::wgpu_core] wgpu-0.19.1/src/backend/wgpu_core.rs:1153: Please report it to https://github.com/gfx-rs/wgpu
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu_hal::gles::device] wgpu-hal-0.19.1/src/gles/device.rs:200: 	Shader compilation failed: 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu_core::device::global] wgpu-core-0.19.0/src/device/global.rs:1602: Device::create_render_pipeline error: Internal error in ShaderStages(VERTEX) shader: 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu::backend::wgpu_core] wgpu-0.19.1/src/backend/wgpu_core.rs:1152: Shader translation error for stage ShaderStages(VERTEX): 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu::backend::wgpu_core] wgpu-0.19.1/src/backend/wgpu_core.rs:1153: Please report it to https://github.com/gfx-rs/wgpu
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu_hal::gles::device] wgpu-hal-0.19.1/src/gles/device.rs:200: 	Shader compilation failed: 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu_core::device::global] wgpu-core-0.19.0/src/device/global.rs:1602: Device::create_render_pipeline error: Internal error in ShaderStages(VERTEX) shader: 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu::backend::wgpu_core] wgpu-0.19.1/src/backend/wgpu_core.rs:1152: Shader translation error for stage ShaderStages(VERTEX): 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu::backend::wgpu_core] wgpu-0.19.1/src/backend/wgpu_core.rs:1153: Please report it to https://github.com/gfx-rs/wgpu
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu_hal::gles::device] wgpu-hal-0.19.1/src/gles/device.rs:200: 	Shader compilation failed: 
imports.wbg.__wbg_warn_b75744a3200b34c1 @ re_viewer.js:2807
re_viewer.js:2807 ERROR: [wgpu_core::device::global] wgpu-core-0.19.0/src/device/global.rs:1602: Device::create_render_pipeline error: Internal error in ShaderStages(VERTEX) shader: 
[... 96 lines cut ...]
    at wasm://wasm/04812d56:wasm-function[4418]:0xa5e839
    at wasm://wasm/04812d56:wasm-function[8568]:0xce8d97
    at wasm://wasm/04812d56:wasm-function[12494]:0xea2308
    at wasm://wasm/04812d56:wasm-function[10281]:0xd8f410
    at wasm://wasm/04812d56:wasm-function[9332]:0xd47a79
    at wasm://wasm/04812d56:wasm-function[7344]:0xc75fc0
    at wasm://wasm/04812d56:wasm-function[947]:0x3f33a7
    at wasm://wasm/04812d56:wasm-function[3914]:0x9d4eef
    at wasm://wasm/04812d56:wasm-function[11562]:0xe3ea58
imports.wbg.__wbg_error_fef71bc9b670aec4 @ re_viewer.js:654
04812d56:0xce8dbd Uncaught RuntimeError: unreachable
    at 04812d56:0xce8dbd
    at 04812d56:0xea2308
    at 04812d56:0xd8f410
    at 04812d56:0xd47a79
    at 04812d56:0xc75fc0
    at 04812d56:0x3f33a7
    at 04812d56:0x9d4eef
    at 04812d56:0xe3ea58
    at 04812d56:0x6cbde3
    at 04812d56:0x582307
index.html:329 Rerun has crashed
check_for_panic @ index.html:329
re_viewer.js:2813 [eframe::web::web_runner] eframe-0.26.0/src/web/web_runner.rs:95: Unsubscribing from 32 events
```

jleibs, 2024-02-06:
Note: can't repro on 0.12.0 so suspect this is related to the the egui / wgpu update rather than another regression in chrome itself.

jleibs, 2024-02-06:
Bisection:
 - Last commit where I cannot reproduce: https://app.rerun.io/commit/5bfb48e/index.html
 - From bd64926 -> 5d9b278 Unknown state since rerun will not load on Chrome (Failed to create wgpu adapter, no suitable adapter found.)
 - First broken commit https://app.rerun.io/commit/7923f49/index.html

jleibs, 2024-02-06:
Also worth noting, forcing `renderer=webgl` doesn't resolve the issue. So it's not related to the try-then-fallback logic.

jleibs, 2024-02-06:
Another observation:
 (1) This is very reminiscent of https://github.com/rerun-io/rerun/issues/3931
 (2) During the range where we weren't able to test on chrome we added https://github.com/rerun-io/rerun/pull/4877 which added just a bit more code to: https://github.com/oxkitsune/rerun/blob/3f787b44ebb4a566224aba821024b27214245575/crates/re_renderer/shader/rectangle_fs.wgsl#L6-L9 which was warned against.

jleibs, 2024-02-06:
Confirmed: reverting https://github.com/rerun-io/rerun/pull/4877 remedies the problem.

emilk, 2024-02-07:
Weird that "Unable to create texture object" is the error message for having a too large shader O.o

Wumpf, 2024-04-09:
@jleibs this got fixed by now?

Wumpf, 2024-10-08:
Since this came down to rectangle shader I'm closing this as a duplicate of

* https://github.com/rerun-io/rerun/issues/3931

closed 2024-10-08 by Wumpf